flathub/org.kicad.KiCad #204: dont use filesystem=home

alexmo1997, 2023-07-29:
Using filesystem=home is not really a good practice as it allows access to the whole user home directory which is not needed here.

jmaibaum, 2023-07-29:
Thanks for opening the issue. I am happy to remove it if it doesn't break the app. Have you already tested how KiCad behaves without the filesystem=home permissions?

If you create a PR removing the permissions, a test build will automatically be created that could easily be tested.

alexmo1997, 2023-07-29:
Weirdly, it doesn't, even though wxwidgets supports using the xdg documents portal afaik. Could be due to a old wxwidgets version or using the *wrong* APIs, that's going to need more investigation.

jmaibaum, 2023-07-29:
I suspect the reason is that a KiCad project consists of multiple files. While you can easily File-Open a project file (using the xdg portal), or double click on a schematic file from you file browser, that doesn't give KiCad permissions to also access the board file, footprints, symbols, and all other project related files needed to work correctly. Those are loaded automatically by the individual tool using traditional file opening calls, which are then blocked as they don't go through the portal.

I am afraid this is not something that we can fix only by altering the flathub flatpak manifest (this repo), but it is a topic that should be brought up with the KiCad main repo at https://gitlab.com/kicad/code/kicad . I'd suggest you file an issue there where it will be seen by the main KiCad Devs.

Yet I don't think that it is an easy to solve task, but I assume it will need major re-enigineering of the whole set of KiCad applications.

It might even be the case that the xdg portal doesn't allow a concept like "auxillary files" next to the file you actually opened through a portal, yet (or never?). I have to admit that I am no expert in this area.

At least in the short term I don't think that we can get rid of the filesystem=home permission. Yet I am leaving this open for further discussion.

jmaibaum, 2023-07-29:
FWIW we are using the most recent stable wxWidgets version, which is 3.2.2.1 at the time of writing this comment.

jmaibaum, 2023-07-29:
I guess the issue we are facing here is this: https://github.com/flatpak/xdg-desktop-portal/issues/463