flathub/org.kicad.KiCad #252: KiCad schematic viewer is extremely slow on asahi fedora

i509VCB, 2023-10-31:
Running on Asahi Fedora 38 (aarch64), the kicad schematic viewer runs extremely slowly when moving the cursor. More than likely hardware acceleration isn't getting enabled.

Please let me know what other information could be useful to diagnose this problem.

i509VCB, 2023-10-31:
Wild guess, kicad seems to use the SDK runtime, is it possible OpenGL is not exposed in the sdk?

https://github.com/flathub/org.kicad.KiCad/blob/9ed573915de60ee4e17d90fc9f134540c8442a89/org.kicad.KiCad.yml#L2C10-L2C29

I wonder if using the org.freedesktop.Platform.GL.default platform would fix this?

jmaibaum, 2023-11-01:
Thanks for your report.

Is only the schematic viewer which runs slowly? Or is pcbnew also slowed down? Have you tried other software from flathub requiring 3D acceleration?

KiCad uses the SDK [since over a year](https://github.com/flathub/org.kicad.KiCad/commit/6d421f23f9e47f411e36c5f6862d18efbd145774) because of the plugin and content manager where pip is available. This allows plugin users to install their favourite plugins' dependencies. Switching back to org.freedesktop.Sdk is thus not an option. If the GL extension would not be available here in general, we would have noticed this. It works as expected on x86_64. If you install KiCad from flathub for the first time the GL extension is pulled as well.

Since the Asahi GPU drivers (and Fedora Asahi as well) are fairly new, my assumption is that the (generic) aarch64 branch of the FDO runtime/SDK is not yet including the necessary plumbing to handle the Asahi drivers.

Unfortunately, I don't have access to a M1/2 Mac running Asahi so I cannot verify my assumption.

jmaibaum, 2023-11-01:
Are you able to launch KiCad's 3D viewer to view a rendering of a PCB? This could be a quick test to check if OpenGL works at all in your flatpak installation of Asahi. I.e. if my assumption about a possible lack of support from the FDO runtime for the Asahi driver is correct or not.

According to KiCad developers it would be unusual, but it could still be that OpenGL works in the 3D viewer, even though Eeschema falls back to software rendering for another reason. So learning about the result of the above test would really be interesting.

i509VCB, 2023-11-01:
I can check some time on the weekend or Friday. I've been switching between the fedora package (crashes due to something related to BTI) and the flatpak since both are buggy in their own ways :)

jmaibaum, 2023-11-01:
I have seen your report on the Fedora bugtracker and the discussion. So the flatpak version does not crash Eeschema? Even more interesting (though completely off topic to this issue).

ninja-, 2024-02-15:
see #276 to get the best possible user experience for now

i509VCB, 2024-02-15:
I actually had a discussion with some people and it's related to asahi not having a stable kernel abi for the gpu yet. This means that the flatpak libraries being used can't do anything on asahi for the moment. I can test #276 but I doubt that is the fix.

ninja-, 2024-02-15:
I see. Well, it would be interesting to know if it even launches to you - it should still be a bit of improvement after dropping XWayland burden

jmaibaum, 2024-02-16:
I think that for Fedora Asahi users the official RPM packages are most likely more performant than a flatpak due to the fact that the GPU drivers do not yet work with the flatpak sandbox.