flathub/org.kicad.KiCad #341: First launch fails except in Gnome

g3r, 2024-11-04:
When I run Kicad for the first time after boot it fails to load, loading it again works as expected.
I can reproduce this in Plasma, sway, hyprland and Cosmic but in Gnome it launches as expected.
If Xwayland is restarted I need to double launch Kicad again.

jmaibaum, 2024-11-06:
Hi, thanks for your report. As I cannot reproduce, possibly because I am using GNOME, I would need your help to circle in on the issue. Do you get any hint what's going wrong running from the terminal, i.e. `flatpak run org.kicad.KiCad`?

g3r, 2024-11-06:
Terminal output is the same on gnome and other compositors:
```
flatpak run --verbose org.kicad.KiCad
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/german/.local/share/flatpak
F: Opening user flatpak installation at path /home/german/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Skipping parental controls check for app/org.kicad.KiCad/x86_64/stable since parental controls are disabled globally
F: Opening user flatpak installation at path /home/german/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.freedesktop.Sdk/x86_64/24.08/9fe67919c01680b69b84980f351d9445606c8493c8355a885169415ca90744c2/files/lib32 does not exist
F: Allocated instance id 2350214161
F: Add defaults in dir /org/kicad/KiCad/
F: Add locks in dir /org/kicad/KiCad/
F: Allowing dri access
F: Allowing homedir access
F: Allowing x11 access
F: Running 'bwrap --args 40 -- xdg-dbus-proxy --args=42'
F: Running 'bwrap --args 40 -- kicad-startup'
Gtk-Message: 12:04:49.975: Failed to load module "colorreload-gtk-module"
Gtk-Message: 12:04:49.976: Failed to load module "window-decorations-gtk-module"
```

jmaibaum, 2024-11-07:
Since there is nothing directly showing up, I would suggest to report this to Xwayland devs as well: https://gitlab.freedesktop.org/xorg/xserver/-/issues

Without any clear hint as to what's going wrong here, there is nothing we in KiCad flatpak can do right now, I am afraid.

jmaibaum, 2025-01-02:
@g3r I see that you have opened https://gitlab.freedesktop.org/xorg/xserver/-/issues/1766 , so I am closing this issue now. Feel free to re-open anew in case it turns out to be an actual issue with KiCad flatpak.

closed 2025-01-02 by jmaibaum