flathub/org.kicad.KiCad #361: Additional KiCad apps not verified on Flathub

kp9001, 2025-02-03:
KiCad is verified on Flathub, and therefore is accessible from the flathub-verified remote. 

```
$ flatpak remote-ls flathub-verified | grep -i kicad
KiCad org.kicad.KiCad 8.0.8 stable
```

However, the various additional apps that are normally installed with KiCad are not verified, and therefore are accessible from the flathub remote but not the flathub-verified remote. 

```
$ flatpak remote-ls flathub | grep -i kicad
KiCad org.kicad.KiCad 8.0.8 stable
Footprints org.kicad.KiCad.Library.Footprints stable
KiCad 3D Model Libraries org.kicad.KiCad.Library.Packages3D 5.1
Packages3D org.kicad.KiCad.Library.Packages3D stable
Symbols org.kicad.KiCad.Library.Symbols stable
Templates org.kicad.KiCad.Library.Templates stable
mariadb-connector-odbc org.kicad.KiCad.ODBCDriver.mariadb-connector-odbc stable
psqlodbc org.kicad.KiCad.ODBCDriver.psqlodbc stable
sqliteodbc org.kicad.KiCad.ODBCDriver.sqliteodbc stable
```

This causes installs from the flathub-verified remote to be broken in some unexpected ways. 

I think this is happening because only the main app has an associated token in the org.flathub.VerifiedApps.txt on kicad.org.  

https://www.kicad.org/.well-known/org.flathub.VerifiedApps.txt

Per the documentation, there should be a unique token for each app. 

https://docs.flathub.org/docs/for-app-authors/verification/#website

imciner2, 2025-02-05:
According to the docs site that you linked:

> I published an extension or a runtime, but the website does not show the verification check
> 
> Verification is only for applications. Extensions or runtimes are not supported.

The only application on the list you provided is the main `org.kicad.KiCad` flatpak, and everything else is an extension. That documentation says verification isn't meant for extensions, therefore we shouldn't add them to that file. If this causes problems with an installation somehow, please contact the FlatPak developers about the problems since it means something else might be going wrong.

jmaibaum, 2025-02-06:
Yes, please raise an issue with the flathub team at https://github.com/flathub/flathub/issues I found this issue https://github.com/flathub/flathub/issues/5117 which was closed after they removed the badge from extensions pages on flathub. They didn't talk about the flathub-verified repo in this issue, so maybe it's indeed an unsolved corner case, even though it surprises me a little bit, since I do not assume that we are the only verified app with extensions.