flathub/org.kicad.KiCad #392: [Bug] Kicad doesn't seem to follow dark mode

K4LCIFER, 2025-03-31:
Despite the theme in my DE being set to dark mode, KiCAD remains in light mode. I see no setting in the preferences to manually toggle this.

DE: KDE Plasma v6.3.3
OS: Arch Linux, Kernel v6.12.21-1-lts

jmaibaum, 2025-04-01:
Thanks for your report. Can you reproduce the issue on non-flatpak KiCad?

K4LCIFER, 2025-04-01:
> […] the issue lies this with Arch's flatpak integration, and not with the KiCad flatpak.

Would it be possible for me to force it to follow the system theme with a flatpak override? I don't want to force any particular theme — I just want it to follow the system theme.

jmaibaum, 2025-04-02:
I am not aware of a flatpak override command regarding dark mode. Since this is an issue with Arch Linux/flatpak/KDE interaction, I suppose you should contact your distro's packaging team about this issue. There is nothing we can do about this from the KiCad flatpak side of things I am afraid

jmaibaum, 2025-04-10:
I'm closing this issue. If there is an actual bug with KiCad flatpak, feel free to open a new issue.

closed 2025-04-10 by jmaibaum