flathub/org.kicad.KiCad #413: Configure update action to run once a week`

bbhtt, 2025-06-07:
I disabled the workflow because it is spamming our build infra with too many PRs. Please configure the workflow to be on a once per week cron

https://github.com/flathub/org.kicad.KiCad/blob/8df5f1fc891c125ba7686d5dde05c29f50da3ca4/.github/workflows/update.yml#L3-L4

bbhtt, 2025-06-07:
It opened 4 PRs in a day all of which are failing https://github.com/flathub/org.kicad.KiCad/pull/412 https://github.com/flathub/org.kicad.KiCad/pull/411 https://github.com/flathub/org.kicad.KiCad/pull/410 https://github.com/flathub/org.kicad.KiCad/pull/409

jmaibaum, 2025-06-08:
Hi, sorry for "spamming," I will definitely improve the pipeline scheduling.

The particular issue in this case is that newly released wxWidgets 3.3 (which is set an "important" dependency) is apparently incompatible with the most recent wxPython release, and I simply did not yet find a free time slot to fix this manually.

I opened an upstream issue about this however (which I am not linking, in order not to spam them).

When I have time later today or tomorrow, I will version-restrict wxWidgets so that that wx 3.3 will not trigger any further builds.

Would you be fine with restricting the pipeline to run once-a-day, instead of once an hour, which was the default I found in the GH Actions of other flathub project when setting it up? Once-a-week would basically force me to do all release updates manually again, because the KiCad release policy is usually releasing an RC (which I deploy via the flathub beta branch), and release the stable version a couple of days later assuming no critical bugs.

If the pipeline only runs once a week, this would mean I have to do updates manually which is cumbersome given that I maintain this entirely in my free time.

F-e-d-c is a great help here combined with automated GH actions, but once-a-week would be so coarse that an entire RC-to-stable cycle could be completed in-between.

I understand that the project is using up build resources with these failing builds. Once-a-day might be a good compromise IMO. What do you think?

bbhtt, 2025-06-08:
Ok

commit fa4bd8b8 (2025-06-09): Run update workflow once per day (3:00 AM)

Fixes: #413
closed 2025-06-09 by jmaibaum