flathub/org.kicad.KiCad #419: Depends on freedesktop SDK at runtime

bkoropoff, 2025-08-09:
The manifest shows the freedesktop SDK as the runtime instead of the freedesktop platform (`org.freedesktop.Platform`):

https://github.com/flathub/org.kicad.KiCad/blob/master/org.kicad.KiCad.yml#L2

This pulls in 600mb of dependencies that are usually only needed for building.  Is it intentional?

jmaibaum, 2025-08-09:
Thanks for the ticket. Yes, this is intentional. The reason is KiCad's plugin manager which allows users to install Python scripts extending KiCad's functionality, and which might need to install Python packages as dependecies.

By depending on the SDK, we greatly simplify KiCad's build process, as the SDK comes with pip and many other things needed, preinstalled.

jmaibaum, 2025-08-22:
Closing as won't fix.

closed 2025-08-22 by jmaibaum